nordicsemi/pc-nrfconnect-ppk #82: how to Measuring current on custom hardware without nRF5 DK(SEGGER J-Link)

amigomcu, 2018-07-06:
hi ,
could be Measuring current custom hardware without nRF5 DK with SEGGER J-Link
i can't see the device(standalone SEGGER J-Link) on ppk 

nRF Connect 2.4.0 on MacOs(10.13.5)

bencefr, 2018-07-06:
Hi @amigomcu , thanks for reporting, we are aware of this issue and we have it on our backlog. Unfortunately to support this feature is a bit complicated, since it involves modifications to several modules and libraries to work correctly.

However in the meanwhile you could try to build the app from this branch: https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/tree/update/core-2.4

there are no guarantees though, and you have to be aware any USB device attach/detach will trigger a device enumeration in nRFConnect which will break the app if a device is currently open.

urish, 2019-01-18:
Hi @bencefr , can you please update on the current status of this?

bencefr, 2019-01-21:
Hi @urish , I'm sorry to inform you that you still need to build the app yourself, however you can use the master branch. I cannot comment about a released version at this time.

urish, 2019-01-21:
Thanks for the update @bencefr. These are good news for me because it means I can actually use and get value out of the product I purchased. 

I tried to code from this repo and was able to get it up and running in 10 minutes, thanks to the instructions in the README, and now I'm finally able to take measurements using the PPK. Thank you!

You may want to update your documentation to reflect this, as I spent some hours looking where I could download the old Python app and trying to get it to work with no success before stumbling upon this repository. 

Also, you may want to update your support team that the possibility of using the code from this repo exists, as they [seem to be unaware](https://devzone.nordicsemi.com/f/nordic-q-a/42663/using-ppk-with-external-j-link-where-to-get-the-software/166461#166461) of this.

chriswils, 2019-03-12:
This feature is now available with the public version. Closing

closed 2019-03-12 by chriswils