nordicsemi/pc-nrfconnect-ppk #100: Connecting to PPK through J-Link makes other J-Link lose target connection

jonas-schievink, 2019-09-20:
I have 2 standalone J-Links connected to my system, one of which is used for the PPK and the other is used to debug my application. If I use the PPK software to connect to the PPK, and then start the JLinkGDBServer to connect to the other J-Link, everything works fine. However, if I do it the other way around, or reconnect to the PPK while the JLinkGDBServer is running, the GDB server will lose the connection to the target and close.

This might well be a bug in the J-Link libraries, in which case feel free to just close this.

jonas-schievink, 2019-09-20:
Ah, this is probably just caused by the target voltage supply from the PPK being interrupted. Still, would be nice if I didn't have to restart the GDB server every time.

wlgrd, 2019-12-09:
So just to be sure, you are powering the other J-Link device with the PPK?

jonas-schievink, 2019-12-09:
The target, yes, both J-Links are USB-powered

wlgrd, 2019-12-09:
If vtarget to jlink is lost, then there are no way of keeping the gdb server as far as I know. The bootup of the ppk needs to turn off the DUT in order to do some offset calculations. Changing this may lead to erroneous measurements.

jonas-schievink, 2019-12-09:
Yeah I thought so, that's unfortunate, but I'll manage.

closed 2019-12-09 by jonas-schievink