nordicsemi/pc-nrfconnect-ppk #99: Renderer process uses a lot of CPU

jonas-schievink, 2019-09-20:
As soon as I connect to a PPK, the renderer process starts using an unreasonable amount of CPU (around 125%). Disconnecting from the PPK again immediately fixes this, so it doesn't seem to be related to any actual rendering.

This is on version 2.2.0 of the PPK software, running on Arch Linux.

wlgrd, 2019-12-10:
This is most probably due to the JLink connection. I have seen other issues with JLink on Linux, and the PPK uses it extensively for communication. Is there any difference when you press "Start" or "Stop", instead of changing connection state?

jonas-schievink, 2019-12-10:
> Is there any difference when you press "Start" or "Stop", instead of changing connection state?

No, not really.

Do you poll the target state in a loop without any delay? I know the `JLinkGDBServer` polls it fairly frequently (and also uses too much CPU for what it does, but that's another issue).

wlgrd, 2019-12-10:
We're using SEGGER RTT read, which I don't know how's implemented on the lower level. All I know is that we continuously poll for incoming RTT data, so that might cause some issues. What specs do you run on?

jonas-schievink, 2019-12-10:
I'm on a Carbon X1 with an i7-8565U @ 1.80GHz

jonas-schievink, 2019-12-10:
I think RTT needs to continuously poll some target memory locations, but that shouldn't require busy-looping like that.

wlgrd, 2020-05-19:
Is this still an issue?

traavnik, 2020-05-19:
Yes. Spawned J-Link process still uses a lot of CPU time. In my case 20-30 % on i7-8550U.
Pressing "stop" does stop RTT traffic according to "J-Link - Web control panel", but CPU usage remains high.

wlgrd, 2020-12-01:
Can you try with the latest release?

traavnik, 2020-12-16:
With the new version (3.0.3), CPU use does decrease to almost 0% when not sampling, which is an improvement.
While sampling using ppk1 it goes up to ~17% on my CPU according to the Task manager. This is perfectly fine as far as I'm concerned.

Here's Process Explorer screenshot with more details in case anyone is interested (PE shows lower CPU usage for some reason):
[image]

wlgrd, 2020-12-17:
Thank you for the update. I will close this, and feel free to add new issues if any.

closed 2020-12-17 by wlgrd