nordicsemi/pc-nrfconnect-ppk #102: Setting VDD does not work consistently/properly

peter-tatch, 2019-10-07:
Unable to  set voltage using regulator. A number of problems with this feature.
Using  NRF Connect v3.2.0 and version 2.2.0 of PPK. 
Using DK for com and power. Power set to regulator, DUT set to external. 

Sliding the value for VDD in voltage regulator control does 1 of a few things. 
1. Does not change the actual voltage to the supply, even though it is updated on the UI
2. Changes to a different value, maybe in the course of sliding
3. Causes hundreds of lines of PPK command failed to be printed to LOG and no change in voltage.

What is going on here? I need to be able to change the voltage and trust that the value on the screen is what the board is actually set to. There are a number of problems in this case as there is no verification that the board is actually set correctly. The PPK command failed printing out hundreds of times in one second makes it seem like there is hundreds of commands being sent at once as the slider slides - may have improper usage of the slider.

The slider as a choice for the UI element is poor to begin with, especially without box for manual setting - no way to easily give exact value to VDD.

Please advise if I am doing something wrong or if this is some known issue/bug. 

See log output below for what I am describing:

2019-10-07T18:23:25.539Z INFO Application data folder: C:\Users\peter\AppData\Roaming\nrfconnect\pc-nrfconnect-ppk
2019-10-07T18:23:25.543Z INFO App initialized
2019-10-07T18:23:25.719Z DEBUG App pc-nrfconnect-ppk v2.2.0 official
2019-10-07T18:23:25.719Z DEBUG App path: C:\Users\peter\.nrfconnect-apps\node_modules\pc-nrfconnect-ppk
2019-10-07T18:23:25.719Z DEBUG nRFConnect 3.2.0 is supported by the app (^3.0.0)
2019-10-07T18:23:25.720Z DEBUG nRFConnect path: C:\Users\peter\AppData\Local\Programs\nrfconnect\resources\app.asar
2019-10-07T18:23:25.720Z DEBUG HomeDir: C:\Users\peter
2019-10-07T18:23:25.720Z DEBUG TmpDir: C:\Users\peter\AppData\Local\Temp
2019-10-07T18:23:25.722Z VERBOSE Could not fetch serial number for serial port at COM1
2019-10-07T18:25:07.067Z INFO Validating firmware for device with s/n 000682514804
2019-10-07T18:25:07.069Z INFO PPK closed
2019-10-07T18:25:07.963Z INFO Opening device with s/n 000682514804
2019-10-07T18:25:07.966Z INFO PPK opened
2019-10-07T18:25:07.967Z INFO Initializing the PPK
2019-10-07T18:25:09.239Z INFO SEGGER serial:  682514804
2019-10-07T18:25:09.240Z INFO SEGGER speed:  2000  kHz
2019-10-07T18:25:09.240Z INFO SEGGER version:  J-Link OB-SAM3U128-V2-NordicSemi compiled Jan  7 2019 14:07:15
2019-10-07T18:25:11.516Z INFO PPK started
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.430Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.431Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.432Z ERROR PPK command failed
2019-10-07T18:29:01.433Z ERROR PPK command failed
[... 265 lines cut ...]
2019-10-07T18:29:01.450Z ERROR PPK command failed
2019-10-07T18:29:01.450Z ERROR PPK command failed
2019-10-07T18:29:01.450Z ERROR PPK command failed
2019-10-07T18:29:01.450Z ERROR PPK command failed
2019-10-07T18:29:01.450Z ERROR PPK command failed
2019-10-07T18:29:01.450Z ERROR PPK command failed
2019-10-07T18:29:01.450Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.451Z ERROR PPK command failed
2019-10-07T18:29:01.452Z ERROR PPK command failed
2019-10-07T18:29:01.452Z ERROR PPK command failed
2019-10-07T18:29:01.452Z ERROR PPK command failed

wlgrd, 2019-10-08:
I know a lot of people are using the regulator, setting it from software, but have never seen this. What is the initial voltage set to? The software shouldn't pass a new value until the slider is released though, so this error tornado should not happen. We'll look into it. 
There have been cases of dropouts when setting the regulator in large steps, so also try and do the change in several, smaller steps.

peter-tatch, 2019-10-08:
The initial voltage is different every time, and doesn't seem to match what is displayed in the window. For example, I reopened the app and found that the sliders is at 3600 mV from what i had before, but the actual output is only 3V. The slider movement causes same command tornado.

peter-tatch, 2019-10-08:
Just closed and reopened and followed your suggestion of moving gradually. Not only did i have to click the slider instead of drag it to move it, but i had to wait between steps otherwise i would get the error. Finally got it up to 3.6 V successfully.

Bottom line is that the slider has got to go. This is way too fickle and confusing on multiple fronts (also can't even choose a value accurately). I suggest a simple number field with increment/decrement button. increment decrement can move in smallest required step, but user should also be able to enter a value and click "set", which will start a process in the background to gradually change the voltage, if this is a true hardware requirement. The UI should indicate when the voltage is in the process of being changed, and when it is at the correct desired voltage.

This could go hand-in-hand with improved process of verifying the voltage on the board. If it helps, just have a value for desired set voltage, and an independent display for actual voltage. That way if there is any discrepancy the user can quickly see and adjust. If the user sets a voltage and software needs to gradually increment it, you should be able to show the actual voltage gradually going up/down towards that value.

wlgrd, 2019-12-10:
"No use and only causes problems" is not accurate in my opinion. It seems that your demands regarding the voltage regulation feature are a bit stricter than the general use cases. The latest release fixes most of your issues. Still, we haven't implemented the feedback you are requesting (but it is way more accurate now). 

Can you please describe your use case, where a gradually set voltage is necessary? Also, the voltage needs to slope to avoid any ESD protection to kick in, but it is now moved from software (sending just one command) to the hardware. A live feedback would choke most of the bandwidth, most probably leading to lost measurement data.

peter-tatch, 2019-12-10:
Hi Wlgrd,

It was you that suggested the gradual voltage change may be important:

"There have been cases of dropouts when setting the regulator in large steps, so also try and do the change in several, smaller steps."

I was referring to that in my suggestion, that if this indeed the case, it shouldn't be the user's job to have to control the rate at which they change the voltage in the UI, this should be controlled by the software. 

I haven't tried the new release yet but will give feedback once I have a chance. I think the main thing is verifying that the voltage displayed on the UI matches the true value on the board, and indicating when this is not the case. Allowing the user to enter or control the precise value for this voltage also seems to be even more important than nice-to-have. 

Let me know if what I meant is still not clear.

Thanks,
Peter

wlgrd, 2019-12-11:
Yes, it's clear. It was never intended for the user to step the voltage in such a way that we avoid a brownout or similar. This bug is now fixed, and the software takes care of stepping through as fast and accurate as it can.
I have taken note of the value input box you are requesting, and keep in mind that the voltage accuracy is now much better. Right now a feedback loop with the exact voltage level is not prioritized, but if we see that the user set value is very much off from the actual voltage output, I will prioritize that. 

Awaiting your feedback, and thank you for reporting :)

wlgrd, 2020-05-19:
Closing, since no further updates.

PR #105 (2019-12-06): Fix updating voltage regulator
This PR fixes #102
closed 2020-05-19 by wlgrd