nordicsemi/pc-nrfconnect-ppk #111: Feature request: Export data

traavnik, 2020-04-09:
As mentioned several times before on Nordic DevZone, it would be very useful to have an option to **export measurement data** from the power profiler application. There have been some indications that the feature is coming soon, but they're now older than one year in some cases.

Very simple CSV dump of current points and a header explaining what the time-step is, would be helpful already. If possible, it would be nice if **only selected data got exported** whenever marker is used.

Your hardware is more than capable for my use cases. I just wish that you got to spend a few more hours on the software side.

Thanks!

h3nrikoo, 2020-05-13:
This would be really beneficial!

traavnik, 2020-05-13:
I recently came across this [ppk_api from wlgrd](https://github.com/wlgrd/ppk_api) which could be useful in a pinch.
Still humbly hoping that someone from Nordic tackles export from the app.

h3nrikoo, 2020-05-13:
Really helpfull @traavnik, thanks a bunch!!

wlgrd, 2020-05-19:
Nice to see that the API helps out. In other news: we are working on a solution for exporting data (thanks to you guys), and the software update will be released soon. I thank you for your patience :)

traavnik, 2020-07-16:
Any update? Thanks

wlgrd, 2020-12-01:
> Any update? Thanks

Export added in latest tag https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/releases/tag/v3.0.0

closed 2020-12-01 by wlgrd