nordicsemi/pc-nrfconnect-ppk #112: Feature request: Snap plot to whole numbers when zooming

traavnik, 2020-04-09:
Zooming on the plots is at the moment completely continuous which leads to some funny axis sizes. There's also a slight cosmetic problem with overlapping numbers at some scales:
[image]

It would be nice if **grid-lines kept snapping to the top and bottom of the plot** (Y axis) while zooming in. Grid generation looks good as is.

For me it's only bothersome on Current axis (Y), which is set to 0-15 mA by default and I spend looking at the uA range most of the time. I'm also almost never interested in negative current, let alone -23,473 uA or whatever I happen to zoom at despite trying my best.

Having and option to export data would alleviate this problem for me, since I would not have to take screenshots to share my results.

Thanks!

wlgrd, 2020-05-19:
We will see if we can get a solution for the zooming. We are currently working on data export #111, so I hope that will help as well.

wlgrd, 2020-12-01:
Fixed in latest tag https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/releases/tag/v3.0.0

closed 2020-12-01 by wlgrd