nordicsemi/pc-nrfconnect-ppk #113: Suggestion: Do not reset Current axis (Y) when going live

traavnik, 2020-04-09:
At the moment any manipulation of Current axis (Y) by holding right click causes the plot to switch to the pause mode. This is fine, but I could not find a way to keep the plot at set scale when going live again (either using "Live" or "Start" button).

Could you make it so that the **scale of the Current axis does not get reset when switching from pause** mode **to live** mode?

You guessed it. This is connected to the other two request. I have to carefully zoom after every new event that I recorded in order to make screenshots with result at least roughly comparable.

Thanks and sorry for nagging this much. Being C focused I don't know anything about javascript and used frameworks. Even trying to bodge some quick fixes would take me days.

enurseitov, 2020-04-22:
+1
I really liked how graph UI was on the old Python version of the PPK (independent per-axis scaling settings(auto-scaling, fixed scaling), markers for averages...). all this is missing in new js version making me stick to the old python-based PPK (which is a nightmare to install on Linux with all it's dependencies).

I also tried to contribute and implement auto-scaling, but man these js frameworks for UI are not for the brains of embedded developers :sweat_smile:

wlgrd, 2020-12-01:
Implemented y-axis lock in latest release https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/releases/tag/v3.0.0

closed 2020-12-01 by wlgrd