nordicsemi/pc-nrfconnect-ppk #114: Power profiler fails to start profiling on MacOS

ball-hayden, 2020-07-28:
Steps to reproduce:

1) Connect an nRF52840DK with power profile shield (PCA63511)
2) Open the power profiler app in nRF Connect Desktop
3) Flash firmware to the power profiler as prompted
4) Observe the PPK is initialised successfully, and "PPK Started" is logged
5) Press the "Start" button

Expected Behaviour:

Profiling starts

Actual Behaviour:

"PPK command failed"

Workaround:

Run profiling tool from inside Windows VirtualBox VM.

wlgrd, 2020-12-01:
Please test this with the latest release of the software :)

wlgrd, 2021-01-08:
Closing due to inactivity. Please reopen the issue if it's still valid.

closed 2021-01-08 by wlgrd

ball-hayden, 2021-01-08:
Sorry @wlgrd - I haven't had chance to test it yet (I don't have a profiler at the moment), but I'll open a new issue if necessary when I have a profiler again.

wlgrd, 2021-01-08:
No problem :)

ball-hayden, 2021-01-27:
@wlgrd I've got hold of a PPK2 (which is way nicer than the v1, by the way) and it's working perfectly now.
Thank you!