nordicsemi/pc-nrfconnect-ppk #141: Incorrect Time Scaling on PPK2

joshbohmer, 2021-01-05:
Time Scaling is incorrect. See attached

Each event shown here is ~40s duration & ~5minute between starts. Power Profiler says 6.34s & 48.8s

[image]
[image]
[image]

wlgrd, 2021-01-05:
Thank you for reporting! Is this happening all the time, also for shorter events? You should see it quite immediately, since this is really a long way off! Also, try and zoom in on the chart and see if there are any data points that seem missing. 
We've seen that if sampling with a large window for a long time, causing the software to display a lot of samples continuously, this behavior might happen (but on a much smaller scale) but haven't been able to reproduce properly. Please also try to sample the same application, but with a smaller window. When you have the data you need, stop sampling and then zoom out to the full view to measure the timing again. 

Also, please post your system specs. Thank you.

joshbohmer, 2021-01-05:
Issue happens immediately on opening the Profiler window & starting the sample. See below.

[image]

In this example 1minute is reported as 10s on the time scale.

I've also tried a small window & same issue as you can see

My system is as follows...
Ryzen 1600, 32GB RAM, 1TB NVME SSD, 8GB 1070, Win 10 Pro

wlgrd, 2021-01-06:
Not good... Can you zoom way in on the chart, like 500µs window size so you see the individual samples? 
[image]
Also try with power off, just for brevity's sake.

joshbohmer, 2021-01-06:
Zoomed In
[image]

Power Off
[image]

Below DUT is a Nordic nRD52 DK w/ nRF Connect Bluetooth Low Energy Firmware as pictured.
[image]
[image]
Same Issue

My Windows 10 region is set to Australia AU in case its a date/time formatting issue.

Requested Below w/logs (power on/off every 2 seconds)
[image]

[2021-01-06T09_04_47.147Z-log.txt](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/files/5774896/2021-01-06T09_04_47.147Z-log.txt)

wlgrd, 2021-01-06:
Thank you. For the feature, please enable the timestamps from the sidebar as well so we can see it along the x-axis:) I see that your zoomed in window measures exactly as expected. Every sample is 10µs, and you mark 12 samples = 120µs. This rules out any date/time formatting issue. I also see that you didn't zoom in on the plot where you power toggle, so I need you to do that, maybe look around a little to see if there are any gaps in the data set. Also, just start sampling and stop some seconds later, so we can correlate the whole range of timestamps along the x-axis with the log.

joshbohmer, 2021-01-06:
Short Sample (10s)
[image]
Log
[2021-01-06T10_44_21.869Z-log.txt](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/files/5775322/2021-01-06T10_44_21.869Z-log.txt)
Data Export
[ppk-20210106T104921_csv.txt](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/files/5775327/ppk-20210106T104921_csv.txt)
Zoomed In
[image]

bencefr, 2021-01-06:
@joshbohmer , the "Short sample (10s)" screenshot does not show the 10s sample in the chart, please navigate the chart to the relevant 10s area. Please note the state of the LIVE VIEW toggle (which is off on the screenshot). If you toggle that on, the chart would be aligned to the latest data. I suggest to first click on the [10s] button, toggle LIVE VIEW on. So far I can't positively judge if there is anything wrong with the app.

joshbohmer, 2021-01-06:
That is the entire chart, 10s is shown as ~1.6s

Same data, with Live View enabled as proof
[image]

wlgrd, 2021-01-06:
Do you observe the incoming data as one to one with the DUT connection (like unplugging, turning it on and off)? Just to know if it is just the time axis that is wrong, but also the timing of the incoming samples.

bencefr, 2021-01-06:
I see. Could you please press Ctrl+Alt+Shift+L, so the Export button changes to Save/Export, then click on it and choose Save. Please share the saved PPK file.

joshbohmer, 2021-01-06:
@wlgrd The data shown on the graph is correct & 1:1 with what's expected. It's just the time scale which shows incorrect & (i think) makes the power usage numbers incorrect. the nRF52 DK should not be using ~10mA

@bencefr Attached
[ppk-20210106T113219.zip](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/files/5775514/ppk-20210106T113219.zip)

joshbohmer, 2021-01-06:
Video Attached, hopefully it helps

https://user-images.githubusercontent.com/3621222/103765355-fad6c000-506c-11eb-8571-c17e27c7458b.mp4

bencefr, 2021-01-06:
There is a massive data loss from the PPK2 to the app somewhere. We have experienced data loss, but not to this nature. Do you have anything unusual in your setup? If you can, please connect the PPK2 to an other, preferably USB3 port. Avoid using HUBs and try changing the cable. Don't use virtual machine environments.

joshbohmer, 2021-01-06:
It plugs directly into a USB3 motherboard port. No hubs. Nothing unusual I can think of.

joshbohmer, 2021-01-06:
Different Cable & Different USB3 Port on the Motherboard

[image]

wlgrd, 2021-01-06:
Can you try in a USB 2.0 port?

joshbohmer, 2021-01-06:
For data loss its VERY consistent. I would expect different results every time if it were data loss. This is always the same. ~6s (Real) = ~1s (Chart)

wlgrd, 2021-01-06:
Also, if possible, please try with a different computer altogether. This is really interesting, and I guess more people will experience a similar issue if this is some hardware compatibility issue. Your findings are appreciated

joshbohmer, 2021-01-06:
So on a USB2 ports I get this... 2s = 1s - Which is an improvement & clearly a hardware (or driver) issue then.

[image]

wlgrd, 2021-01-08:
Thank you for your reports. We will work to reproduce this locally and get back to you.

wlgrd, 2021-01-14:
Can you share information about your chipset?

wlgrd, 2021-02-15:
@joshbohmer Just bumping to get some information about your chipset. Also, please check if 3.1 solves this issue for you.

kareszv13, 2021-03-03:
I have similar issue (v3.1.0)

# nRFConnect System Report - 2021-03-03T12-46-14.367Z

- System:     Gigabyte Technology Co., Ltd. B450M S2H
- BIOS:       American Megatrends Inc. ALASKA - 1072009
- CPU:        1 x AMD Ryzen 7 2700X Eight-Core Processor 3.7 GHz 16 cores (8 physical)
- Memory:     6.6 GB free of 15.9 GB total
- Filesystem: C: (NTFS) 418.4 GB 88.2% used

- OS:         Microsoft Windows 10 Education (10.0.19042) win32 ia32

- Versions
    - kernel: 10.0.19042
    - git: 2.24.1.windows.2
    - node: 12.13.0
    - python: 3.8.2
    - python3: 

- Connected devices:
    - COM3: C43C9BA7C356

rodrigobrochado, 2022-01-31:
I'm having the same problem with a PPK (old one) attached on top of a nrf52840-DK. I have already tested with different versions for nRF connect (3.9.3, 3.71 and 3.6.1) and Power Profiler app, and flashed different firmwares on the PPK nrf52832 using those versions. Also tested on Ubuntu 21.10, 20.10 (it has already worked on 20.10 before) and on a different notebook with Windows 10.

I'm kind of skeptical, but due the age of both boards, could this problem be caused by hardware damage? @wlgrd

wlgrd, 2022-02-01:
So you have incorrect time scaling as well? That's an issue we haven't seen with the PPK, only the PPK2. The time scale should definitely not be affected by the age of the hardware.

rodrigobrochado, 2022-02-01:
Yes, incorrect time scaling. I'm also not getting correct and reproducible measurements values, that's why i though of hardware damage. e.g.: I normally get constant 3mA for an erased nrf52840 chip, but I'm getting some strange behavior:
[image]

Sometimes it even gets constant 25 uA for the erased chip...

rodrigobrochado, 2022-03-04:
New PPK2 working fine on same PC and settings (had just to update [udev rules](https://github.com/NordicSemiconductor/nrf-udev)). It seems that both our old ppk and nrf DK was not working as expected. I would blame sea air corrosion...

wlgrd, 2022-03-07:
Strange indeed. Can you give me the label information of both the working and failing PPK2's that you have?

rodrigobrochado, 2022-03-07:
The failing one is a PPK1, not PPK2, mounted on top of a DK. 
PPK1 label: PCA63511 Rev 1.1.0, 2018.44
DK label: PCA10056 1.1.0, 2019.11

wlgrd, 2022-03-07:
Ah, of course. Sorry for that. It should definitely not be caused by aging hardware at least. For the PPK1 case, there are aspects of it that we cannot control due to using the debugger firmware from SEGGER to provide the transport  of the data. Will keep an eye out for more cases, so thank you for reporting it.

wlgrd, 2022-03-07:
Since OP hasn't responded in some time and others reporting PPK2 working fine now, I will close this. Please open another issue if this occurs again.

closed 2022-03-07 by wlgrd