nordicsemi/pc-nrfconnect-ppk #159: Digital channels do not update

bobatsar, 2021-01-13:
I am trying the digital channels and it seems they stay at always the same level.

I am working in Source meter Mode and am running PPK2 with v3.0.3 on Ubuntu 20.04.

For testing I connected different logic port channels to either VCC or GND outputs of the profiler (I tried also with the logic port outputs). The level of the ports in the GUI stays always at the same level.

Sometimes at startup the levels are at 1, sometimes at startup the levels are at 0.
All channels have the same level.

I saw there were some commits lately about digital channels. Could this already be fixed? Can you make a new release?

wlgrd, 2021-01-14:
You say you have tested connecting logic port channels to VCC or GND: these are not outputs, but inputs. You should have your device's VDD and GND connected here. The logic port of the PPK2 has a level shifter that needs to know your DUT's digital voltage level to be able to safely translate them to the PPK2's MCU. 
 
Your logic port channels, what are they driven by, at which voltages? Can you also verify the voltage of your outputs, when connected to the Dx pins of the PPK2? 

A quick test you can do is when in SMU mode, just connect VOUT and GND from the 4-pin cable to VCC and GND on the logic port connector. Then, take a wire and short between VCC to a Dx pin, then GND to the Dx pin. That way you can verify the functionality of the port.

The commits are related to the graphical representation, and has nothing do to with the functionality of the port :)

bobatsar, 2021-01-15:
Hi @wlgrd,

yes you were right, it seems I didn't RTFM and thought they were outputs.
But it makes totally sense to have these as inputs with a level shifter.

Now everything is working.

Thanks, but I am also waiting for the graphical representation improvements. 30s for digital inputs will be much better than 3s :-)

closed 2021-01-15 by bobatsar

wlgrd, 2021-01-19:
I agree :) Thank you for coming back and closing the issue.