nordicsemi/pc-nrfconnect-ppk #190: feature requests

pabigot, 2021-02-13:
I'm finding this with PPK2 much more useful than the old PPK and UI, but there are a few things that would be very useful, as of 3.0.3 which is the version available to me.

* Y axis lock is nice, but being able to set it explicitly is pretty important.  Now if I want to cap at 4 mA I have to take a sample, manipulate the display until the auto-adjusted Y axis is about right, then lock it.  Which works, but if I then take a sample and need to zoom in on low-power, I won't be able to get back the original scaling.  Which means images captured from successive sessions will have different axes.
* It follows that explicit setting of the X axis would also be helpful.
* The statistics boxes are a start, but the minimum value in the window should also be provided along with the max.  In my experience the max is often from noise spikes, which biases the average, which means I don't have a valid number to represent best-case power usage in a particular mode.  A median filter or histogram could also be helpful.
* Ability to export is somewhat useful, but ability to import back would be even more useful, since otherwise I'd have to find another time series analysis UI to go back and look at previous results.  (*Really* helpful would be ability to have multiple windows, so we can compare captures by eye.)
* I'd love a way to trim a capture to an area of interest.
* I was going to ask for trigger, but it looks like that'll be in 3.1.0.  That's not apparently packaged yet, but if it doesn't also add support for setting duration of capture I'd like that added: for the analysis I'm doing today I want to start as soon as power consumption goes above 1 uA and capture the next 10.5 s.  (Ideally the trigger could be either from a logic signal or from a power threshold.)

Also if you could add basic instructions on how to build and run this from the source directory, if that's at all possible, then I might be able to contribute to some of these features.

Thanks for your work on this.

nicolai-nordic, 2021-02-15:
Hello @pabigot and thanks for some great suggestions! We just released 3.1 which includes some of these, and I will add the others to our backlog.

A couple of comments/questions:
* With the release of 3.1 it is now possible to import a previously exported data set (and you can open multiple instances of the app to compare captures).
* Can you describe in more detail what you mean by trimming a capture to an area of interest?

Regarding app development, you can take a look at this guide:
https://nordicsemiconductor.github.io/pc-nrfconnect-docs/app_development

resghst, 2022-03-13:
I have a suggestion for feature: 
In export CSV timestamp, the timestamp may need more format. For example, now the timestamp exportation is from 0 to the end time, but I want to know this start time can map to PC time. It will help more applications to know the power information.

Thanks, NordicSemiconductor providing many powerful tools for us.

rei-vilo, 2022-04-15:
I really enjoy the PPK2 with software release 3.4.3.

However, I couldn't find how to use a trigger mentioned earlier.

> * I was going to ask for trigger, but it looks like that'll be in 3.1.0.  That's not apparently packaged yet, but if it doesn't also add support for setting duration of capture I'd like that added: [...]

The two options (current threshold and logic signals) are great.

Thank you!

boaks, 2022-07-22:
> Can you describe in more detail what you mean by trimming a capture to an area of interest?

I'm not sure, what originally was meant. I would like to have a function, which zooms the "selected range" (y-axis, area of interest):

[image]

that would then show the selected range (21s) extended (zoomed) to the "complete y-axis".
(Hope, it's clear enough :-) ).

pabigot, 2022-07-22:
What I meant was an ability to select a range as @boaks showed, and then discard all data outside that range so that further exploration and exported data wouldn't have the readings I didn't care about.

homonto, 2022-08-07:
> https://nordicsemiconductor.github.io/pc-nrfconnect-docs/app_development

100% needed - I am with you.
Especially on laptop, when you have no mouse but must use the trackpad only - I would see it this way:

1. select the time frame
2. click: Auto-zoom
3. say thank you to developers ;-)
please

tomav, 2023-01-04:
Same as @resghst, other timestamp/date formats will be appreciated when trying to compare consumption with log files from a device. At least Unix timestamp with milli/micro second.

boaks, 2023-01-04:
@tomav 

That would require to sync the times from the device with those on the PC.
I use an other approach with additional GPIO, see [zephyr coaps client / powerconsumption](https://github.com/boaks/zephyr-coaps-client/blob/main/docu/POWERCONSUMPTION.md#nrf9160-dk-nb-iot-with-psm-and-rai). The signal 1 correlates to the RRC connection state.

aadnekar, 2023-03-08:
@boaks, would this PR be what you are interested in? I've included a video in the PR.
https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/pull/310

homonto, 2023-03-08:
> @boaks, would this PR be what you are interested in? I've included a video in the PR. #310

man, zoom selection IS THE GOLD if implemented ;-)

boaks, 2023-03-08:
Great! Yes!

boaks, 2023-03-09:
> man, zoom selection IS THE GOLD if implemented ;-)

Yep, it's merged :-). 
It's not too hard to build it locally (it's mainly a question to chose the right node.js version. For the Programmer it's v18, for the Power Profiler that seems to be too new but also works).
I guess, the next release will also come.

boaks, 2023-10-03:
Any plans, when this will be included in a release?