nordicsemi/pc-nrfconnect-ppk #191: Set VDD feature is not working at all - PPK

selman-nus, 2021-03-08:
Hi,

It was very disappointing to observe this error because I have submitted a report based on PPK results. 
.
I have just checked whether the VDD set feature of the PPK is working correct or not. I adjusted DUT to external and used my fluke multimeter to measure the output from the External DUT header. Readings were always 2.785 volts no matter what I did. You can see the log below.
.
All the software is up to date and I checked every hardware connection twice. 
.
Is this same for everyone? Is it possible to fix this with a software update or is this hardware related issue?

```
2021-03-08T10:46:37.087Z INFO Validating firmware for device with s/n 000682699356
2021-03-08T10:46:37.125Z INFO PPK closed
2021-03-08T10:46:37.578Z INFO Opening device with s/n 000682699356
2021-03-08T10:46:37.598Z INFO Initializing the PPK
2021-03-08T10:46:37.929Z INFO SEGGER serial number:  682699356
2021-03-08T10:46:37.929Z INFO SEGGER speed:  2000  kHz
2021-03-08T10:46:37.929Z INFO SEGGER version:  J-Link OB-SAM3U128-V2-NordicSemi compiled Mar 17 2020 14:43:00
2021-03-08T10:46:39.350Z INFO PPK started
2021-03-08T10:46:39.358Z INFO PPK opened
2021-03-08T10:47:13.157Z INFO Voltage regulator updated to 1980 mV
2021-03-08T10:47:15.693Z INFO Voltage regulator updated to 2041 mV
2021-03-08T10:47:18.752Z INFO Voltage regulator updated to 2916 mV
2021-03-08T10:47:21.933Z INFO Voltage regulator updated to 2838 mV
2021-03-08T10:47:23.566Z INFO Voltage regulator updated to 2803 mV
2021-03-08T10:47:26.703Z INFO Voltage regulator updated to 2725 mV
2021-03-08T10:47:52.389Z INFO Voltage regulator updated to 1850 mV
2021-03-08T10:47:55.933Z INFO Voltage regulator updated to 1963 mV
2021-03-08T10:47:57.934Z INFO Voltage regulator updated to 2396 mV
2021-03-08T10:47:59.805Z INFO Voltage regulator updated to 3600 mV
2021-03-08T10:48:14.382Z INFO Voltage regulator updated to 1900 mV
2021-03-08T10:48:29.607Z INFO Voltage regulator updated to 1950 mV
2021-03-08T10:48:35.752Z INFO Voltage regulator updated to 1950 mV
2021-03-08T10:48:35.842Z INFO DUT OFF
2021-03-08T10:48:36.998Z INFO DUT ON
2021-03-08T10:48:39.044Z INFO Sampling started
2021-03-08T10:48:41.952Z INFO DUT OFF
2021-03-08T10:48:43.433Z INFO DUT ON
2021-03-08T10:48:46.322Z INFO Voltage regulator updated to 2288 mV
2021-03-08T10:48:49.712Z INFO DUT OFF
2021-03-08T10:48:50.468Z INFO Sampling stopped
2021-03-08T10:48:52.306Z INFO Sampling started
2021-03-08T10:48:53.183Z INFO DUT ON
2021-03-08T10:48:54.505Z INFO Voltage regulator updated to 2505 mV
2021-03-08T10:48:57.239Z INFO Sampling stopped
2021-03-08T10:48:58.013Z INFO DUT OFF

```
I read PCA63511 on my PPK board. Rev 1.1.0 and date is 2018.26.

[image]

wlgrd, 2021-03-08:
It sounds to me like you have the `Power Select` switch set to `DK`, and not `Reg`. Can you verify this? If set to `DK`, then the regulator setting will do absolutely nothing. I am testing with the latest software, and it's working correctly. 
That being said, the output voltage is not completely accurate, so it's smart to verify with your voltmeter if you need a very accurate voltage setting.

selman-nus, 2021-03-09:
Ups, I am so sorry, I haven't realized the power select switch was important. I tested with that config and it works as expected. Thanks for your fast advice!

closed 2021-03-09 by selman-nus

wlgrd, 2021-03-10:
No problem, glad it worked out for you. And indeed, the amount of analog switch setting was something we worked hard to get rid of for the PPK2 since it might be a source of confusion and easy mistakes.