nordicsemi/pc-nrfconnect-ppk #198: Chart issues when loading from saved capture file

devanlai, 2021-06-01:
Version: 3.1.1

When I load a previously saved `.ppk` file, the entire capture is initially displayed, but as soon as I zoom or pan, the chart zoom extents reset to a maximum of 2 minutes and the time span selections in the top bar are limited to at most 1 minute. 

Additionally, it may initially display the "LOCK Y-AXIS" toggle as enabled, but it doesn't actually lock the Y-axis until I toggle it off and on again.

From poking at the internals, it appears that the source of the zoom issues is that the chart extents are tied to the internal `app.dataLogger` / `dataLoggerState`, which do not seem to be updated when loading a file.

When it attempts to calculate which time spans to show in the top bar in [ChartTop.jsx](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/blob/master/src/components/Chart/ChartTop.jsx#L77), it thus appears to be calculating the span using the defaulted max sample rate of 100kHz, so it selects the first six time spans.

Similarly, in [chartReducer.js](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/blob/180a4e5773fca4889bc3a10bbd3946375b48e43a/src/reducers/chartReducer.js#L101), it uses the defaulted max sample rate when calculating the maximum window time span of 120 seconds.

Ideally, the chart should be able to correctly update the available time span zoom levels based on the actual duration of the loaded capture. This would make it much easier to go back and review previous captures. Right now, if I want to view more than two minutes at a time I have to do one of the following workarounds:
1. Look at it during the initial capture
2. Export the data to a CSV file
3. Patch the PPK app

nicolai-nordic, 2021-06-08:
Hello @devanlai! Thanks for reporting this issue, we will look into it.

devanlai, 2021-06-09:
From testing, #199 seems to fix the issue for any new capture files saved with v3.1.2. I'm looking forward to seeing v3.1.2 officially released.

nicolai-nordic, 2021-06-14:
v3.1.2 has been officially released and should hopefully fix this issue. Let us know if it does not work as intended. Again, thanks for reporting it.

closed 2021-06-14 by nicolai-nordic