nordicsemi/pc-nrfconnect-ppk #202: Empty export

DaveDavenport, 2021-06-16:
I am trying to make a trace and save it. But when I export I get a file with only a header:
```
19:16:02.354	Exported CSV to: /home/e/ppk-20210616T171556.csv
19:17:23.520	Export cancelled
19:17:23.531	Exported CSV to: /home/e/ppk-20210616T171719.csv
19:17:51.183	Export cancelled
19:17:51.193	Exported CSV to: /home/e/ppk-20210616T171746.csv
```

The files are empty except header.
```csv
Timestamp(ms),Current(uA)
```

I am on version 3.1.2 (Official).

```
# nRFConnect System Report - 2021-06-16T17-27-46.490Z

- CPU:        1 x Intel® Core™ i7-6700HQ 2.60 GHz 8 cores (4 physical)
- Memory:     573.3 MB free of 15.6 GB total
- Filesystem: /dev/sda2 (ext4) 233.2 GB 93.6% used

- OS:         Ubuntu (20.04.2 LTS) linux x64

- Versions
    - kernel: 5.4.0-74-generic
    - git: 2.25.1
    - node: 12.13.0
    - python: 3.8.5
    - python3: 3.8.5

- Connected devices:
    - /dev/ttyACM0: 
  ```

nicolai-nordic, 2021-06-16:
Does this happen every time you try? Do you have any other computers available to see if you see this problem there also?

DaveDavenport, 2021-06-16:
After closing/reopening a few times, it seems that sometimes it works.
Ill try another pc tomorrow.

DaveDavenport, 2021-06-24:
cannot reproduce on other machines.

closed 2021-06-24 by DaveDavenport