nordicsemi/pc-nrfconnect-ppk #216: Cannot detect the serial ports of PPK (v3.2.0)

DavidKimWellysis, 2021-11-04:
After v3.2.0 Today, cannot detect the serial ports of PPK

Although reboot computer or reinstall software, the same problem is occurring.

[image]

DavidKimWellysis, 2021-11-04:
Rolling back to nrf connect 3.7.1, upgrading to 3.8.0 solved the problem.

closed 2021-11-04 by DavidKimWellysis

datenreisender, 2021-11-04:
Glad to hear, that your problem is solved. 

Sadly, we hear from users that this problem occasionally (re)occurs. For some, a simple restart of nRF Connect for Desktop also solved the problem, but for some it is pretty persistent. So, we are working on a more thorough solution. 

FYI: If the problem reappears or is more persistent, downgrading to 3.7. might be a workaround, until we have a better solution. Alas, downgrading is not trivial since the new versions of the app do not run with 3.7 any longer. So if you happen to want to do this, this is the way to downgrade:

1. Download the latest 3.7 version of nRF Connect for Desktop from https://www.nordicsemi.com/Products/Development-tools/nRF-Connect-for-desktop/Download?lang=en#infotabs (for macOS this is 3.7.2, for Windows and Linux 3.7.1, because 3.7.2 was not needed and not released for these)
2. After starting nRF Connect for Desktop 3.7 a prompt will come up about the 3.8 upgrade. Click “No” on this.
3. In the Settings, click on “Add source”, enter the URL http://developer.nordicsemi.com/.pc-tools/nrfconnect-apps/3.7-apps/apps.json and click “Add”.
4. Back in the list of all apps, all apps should show up twice now, one with “official” below them and one with “3.7 compatible apps” below them. If you want to hide the incompatible versions, click on “Filter” and deselect “Official”.
5. Install the apps you need in the old versions and use them.

DavidKimWellysis, 2021-11-05:
Thank you for your guide, kindly.

I solved the issue.
I hope to use your official service ASAP.

Sincerly.
David Kim
________________________________
보낸 사람: Marko Schulz ***@***.***>
보낸 날짜: 2021년 11월 5일 금요일 오전 12:21
받는 사람: NordicSemiconductor/pc-nrfconnect-ppk ***@***.***>
참조: David Kim ***@***.***>; State change ***@***.***>
제목: Re: [NordicSemiconductor/pc-nrfconnect-ppk] Cannot detect the serial ports of PPK (v3.2.0) (Issue #216)

Glad to hear, that your problem is solved.

Sadly, we hear from users that this problem occasionally (re)occurs. For some, a simple restart of nRF Connect for Desktop also solved the problem, but for some it is pretty persistent. So, we are working on a more thorough solution.

FYI: If the problem reappears or is more persistent, downgrading to 3.7. might be a workaround, until we have a better solution. Alas, downgrading is not trivial since the new versions of the app do not run with 3.7 any longer. So if you happen to want to do this, this is the way to downgrade:

  1.  Download the latest 3.7 version of nRF Connect for Desktop from https://www.nordicsemi.com/Products/Development-tools/nRF-Connect-for-desktop/Download?lang=en#infotabs (for macOS this is 3.7.2, for Windows and Linux 3.7.1, because 3.7.2 was not needed and not released for these)
  2.  After starting nRF Connect for Desktop 3.7 a prompt will come up about the 3.8 upgrade. Click “No” on this.
  3.  In the Settings, click on “Add source”, enter the URL http://developer.nordicsemi.com/.pc-tools/nrfconnect-apps/3.7-apps/apps.json and click “Add”.
  4.  Back in the list of all apps, all apps should show up twice now, one with “official” below them and one with “3.7 compatible apps” below them. If you want to hide the incompatible versions, click on “Filter” and deselect “Official”.
  5.  Install the apps you need in the old versions and use them.