nordicsemi/pc-nrfconnect-ppk #231: Power profiler fails on first connection to PPK

acanidio, 2021-12-15:
Corrupted data is received when first connecting to PPK, after opening the application.

### Setup:
- PPK+DK52 powering an external DUT
- Windows 10 x64
- Power Profile v3.2.1

### Issue

The application reports that corrupted data has been detected. To allow power profiling it is needed to reconnect to the board.

[image]

Attached also the application log to show the behavior: [2021-12-15T09_17_04.951Z-log.txt](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/files/7717986/2021-12-15T09_17_04.951Z-log.txt)

Available to add further details if needed.

PS: I am thinking to buy the PPK2 but I am not quite sure because the with PPK these bugs are quite annoying.

nicolai-nordic, 2021-12-15:
Just to address your concern regarding buying a PPK2. In my experience at least it is vastly more stable than the PPK1, and offers more functionality, with even more coming in the future. I would recommend at least trying it to see if it fits your requirements. 

Now back to your issue. Is this something that started happening after updating the PPK app or nRF Connect for Desktop? If so, can you try downgrading to nRF Connect for Desktop 3.7.1 with the associated PPK app (3.1.3 I believe). There was some changes made to how we communicate with the PPK1 in the nRF Connect For Desktop 3.8 release. These are the instructions for downgrading:

```
1. Download the latest 3.7 version of nRF Connect for Desktop from https://www.nordicsemi.com/Products/Development-tools/nRF-Connect-for-desktop/Download?lang=en#infotabs (for macOS this is 3.7.2, for Windows and Linux 3.7.1, because 3.7.2 was not needed and not released for these).
2. After starting nRF Connect for Desktop 3.7 a prompt will come up about the 3.8 upgrade. Click “No” on this.
3. In the Settings, click on “Add source”, enter the URL http://developer.nordicsemi.com/.pc-tools/nrfconnect-apps/3.7-apps/apps.json and click “Add”.
4. Back in the list of all apps, all apps should show up twice now, one with “official” below them and one with “3.7 compatible apps” below them. If you want to hide the incompatible versions, click on “Filter” and deselect “Official”.
5. Install the apps you need in the old versions and use them.
```

Also are you able to get measurement data at all or does it never work?

acanidio, 2021-12-18:
It happened after 3.7. 1 update. I will try to downgrade to see if the issue is not present there and I will let you know soon.

acanidio, 2021-12-21:
I downgraded to version 3.7.1 and it seems to me that is far more stable but the issue aforementioned is again there even if it is slightly different.

Here I attached the log file: [2021-12-21T09_31_35.096Z-log.txt](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/files/7751186/2021-12-21T09_31_35.096Z-log.txt)

Hope to hear from you soon.

acanidio, 2022-01-13:
@nicolai-nordic any news about the described issue? Does any of the new nRF Connect 3.9.3 address the described issue, since I want to update but I cannot afford to try and revert any new update

nicolai-nordic, 2022-01-14:
Sorry for not replying to your previous comment @acanidio. I unfortunately have no news on the PPK1, as this is not something we have been able to prioritize. So 3.9.3 will likely not address the issue you describe.

Perhaps @wlgrd has any insight into why you are getting corrupt samples on your PPK1?

wlgrd, 2022-01-14:
Which version of the SEGGER JLink are you at? They have introduced, removed, and reintroduced bugs from time to time with regards to RTT/UART transfers. Downgrading nRF Connect for Desktop doesn't downgrade the SEGGER binaries. Please verify your current version, and also update to the latest manually from https://www.segger.com/downloads/jlink/ and see if that helps

acanidio, 2022-01-24:
On my system i have installed:
- J-Link v6.94a
- J-Link v7.50a
- J-Link v7.60e

I noticed that Power Profiler is using J-Link v6.94a, how can I change the J-Link version used by nRF Connect?

wlgrd, 2022-01-25:
From what I understand, this is bundled together with the nRF Connect for Desktop application. @nicolai-nordic please correct me if I am wrong here. 
Is the instability you are now seeing something you have had before the upgrade as well? Also, please connect the DK you are using, open J-Link Configurator and verify that the version running on the debugger is up to date (or the same as the software is running). The 52DK is quite old, and if it hasn't automatically updated then that can also be an issue. 

I am sorry for the inconvenience as this was one of the main reasons for doing PPK2 without using RTT/Debugger as the communication channel. Being that the J-Link firmware is out of our control and from time to time a bit unstable for this use, we have seen this issue previously as well without any clear solution.

nicolai-nordic, 2022-01-27:
It is bundled with nRF Connect for Desktop, but we do not enforce the usage of this particular version. Currently we recommend using 7.58b, which is also the one bundled with nRF Command Line Tools v10.15.1. I'm not sure if there are any better ways to use this version than to uninstall other older versions of JLink.

acanidio, 2022-01-31:
Thanks for the advice. I manage to downgrade the debugger firmware and now it all works. The corrupted data issue is still there but I have ordered a PPK2 and I will have it soon, so hopefully this will no longer be a problem.

closed 2022-01-31 by acanidio