nordicsemi/pc-nrfconnect-ppk #241: COM port is busy (access denied) or generally just doesn't work

mortenmoulder, 2022-01-14:
I'm running Windows 11 and I have been experiencing some weird issues with my PPK2. Once in a while it works flawlessly, as in no issues what so ever. Then suddenly after opening the Power Profiler app, it simply will not connect to the device. I can see it in the list, but it never connects.

I read multiple places that I should try to downgrade to 3.7.1. That did not solve it either.

I also read people saying, that if I open the Programmer first, put it into bootloader mode, and then open Power Profiler, I should accept the firmware install. That has surprisingly worked a few times.

If I open Programmer, put it into bootloader mode, open Power Profiler and proceed with the firmware install, I get an "Error while setting up device XXXXXX: Opening COM4: Access denied". I tried as administrator, but to no avail.

If I use [Microsoft's Process Explorer](https://docs.microsoft.com/da-dk/sysinternals/downloads/process-explorer), and I search for `usbser` using [this guide](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA03q000000YGw9CAG&l=en-US), I get 0 results back, which probably means the COM port isn't being used by another application.

If I unplug it, plug it back in, it's now COM3 because it's out of the bootloader again. If I open Power Profiler and select the device, I just get this in the log and nothing else:

```
2022-01-14T00:16:51.887Z INFO Application data folder: C:\Users\morte\AppData\Roaming\nrfconnect\pc-nrfconnect-ppk
2022-01-14T00:16:51.953Z DEBUG App pc-nrfconnect-ppk v3.1.3 official
2022-01-14T00:16:51.953Z DEBUG App path: C:\Users\morte\.nrfconnect-apps\external\3.7 compatible apps\node_modules\pc-nrfconnect-ppk
2022-01-14T00:16:51.953Z DEBUG nRFConnect 3.7.1 is not supported by the app (^3.6.1)
2022-01-14T00:16:51.953Z DEBUG nRFConnect path: C:\Users\morte\AppData\Local\Programs\nrfconnect\resources\app.asar
2022-01-14T00:16:51.953Z DEBUG HomeDir: C:\Users\morte
2022-01-14T00:16:51.954Z DEBUG TmpDir: C:\Users\morte\AppData\Local\Temp
2022-01-14T00:16:53.608Z INFO Validating firmware for device with s/n XXXXXX
2022-01-14T00:16:53.619Z INFO Opening device with s/n XXXXXX
```

Completely dead. If I reboot my PC, it's most likely going to work again, but that's a huge pain in the rear to do every time.

Is this a Windows problem?

nicolai-nordic, 2022-02-01:
Hello @mortenmoulder. Sorry it took so long to get back to you. I haven't seen this behavior before, so I'm not really sure what the best way to proceed with this is.

We are in the process of releasing nRF Connect for Desktop 3.10, which will include the option of turning on verbose logging for nrf-device-lib, i.e. the underlying library handling device operations. Enabling verbose logging while the device is in this unselectable state might give us some useful information.

As I write this we are doing the final testing stages on 3.10, so hopefully it will be available within a few days.

mortenmoulder, 2022-02-01:
Hi @nicolai-nordic - thanks for the update. I'll give it a go when it's released.

In the meantime, you might want to try to install WSL (Debian for example) on a Windows PC. I've been consistently experiencing these issues, and it's only when I launch Debian through WSL that the device doesn't work anymore. You can install Debian through the Microsoft Store.

nicolai-nordic, 2022-02-01:
I had WSL with Ubuntu installed and tried launching that, but it didn't happen. I'm not on Windows 11 yet though, so I will look into this.

mortenmoulder, 2022-02-01:
Aha, hmmm. That might be a Windows 11 issue then!

mortenmoulder, 2022-04-09:
Just checking in to say, that this issue is still happening and it's VERY annoying. Have to reboot my PC every time I want to use my PPK2.

wlgrd, 2022-04-11:
To me it sounds like Windows is reporting the PPK2 as available up to the software. We have never seen this on any Linux distribution, Mac OSX or Windows 10. Are you able to test this on another computer, not using Windows 11?

mortenmoulder, 2022-04-11:
@wlgrd I don't think that's the right approach, to be fair. Let's say I cannot replicate on Windows 10, then nothing happens.

I'm sure there are people working at Nordic Semiconductor who has a PPK2 and who runs Windows 11. Installing WSL through Microsoft Store takes ~3 minutes.

wlgrd, 2022-04-11:
This is to ensure that the software and your PPK2 are working as it should and that W11 is indeed the culprit. I know how flaky the use of serial ports can be with Windows, and adding WSL into the mix makes it even worse and will not only affect the PPK2 for sure. Can you confirm that this is only happening when WSL is running at the same time?

mortenmoulder, 2022-04-11:
@wlgrd I can confirm that if I have PPK2 connected and I'm monitoring, if I open WSL it immediately disconnects and is unable to connect to PPK2 until I reboot.

I'm not even doing anything with WSL. I just open the Debian terminal and that's it.

I will test on a Windows 10 machine later, but thus far I have not been able to replicate it on Windows 10. Luckily I have 2x Windows 11 and 1x Windows 10 machine at home, so I should be able to confirm it's a Windows 11 issue.

mortenmoulder, 2022-04-11:
@wlgrd Can confirm that it is not happening in Windows 10, but it's happening on two Windows 11 machines with Debian installed in WSL.

I have not tried a fresh install and just installed the Power Profiler and Debian in WSL, but I think it's safe to assume, that the issue is happening in Windows 11.

However, all of my other serial ports work. I even tried flashing a pretty large firmware to an Arduino, then opened WSL, and it didn't stop flashing.

Anyone I can reach out to from Nordic Semiconductor, who can work on this issue?

mortenmoulder, 2022-04-19:
@wlgrd Our entire enterprise of 1000+ people are using Windows 11. It came out half a year ago, so I think Windows 11 support is a requirement (with WSL, which is also common for the likes of us).

jonasem, 2022-04-20:
I tried this now on Windows 11, Ubuntu WSL2, PPK v3.4.3, nRF for Desktop v3.11.0 and could not reproduce. I even flashed the ppk with some other fw and flashed it again with the bundled fw from the ppk app and it just works. 

Is there anything else that is done with your WSL except from installing ubuntu from the store? Trying to reproduce as much as possible.

I haven't used the launcher version reported 3.7.1, but the most current one, 3.11.0, maybe there is something in between there that fixes it? Have you tried recently with a current version of nRT Connect for Desktop?

jonasem, 2022-04-20:
I downgraded to nRF Connect 3.7.1 and ppk v3.1.3, and it still works fine, launching both Ubuntu and Debian mid-trace.

mortenmoulder, 2022-04-20:
@jonasem Thank you for trying!

Try launching WSL via Windows Terminal (it's also in the Microsoft Store). I can replicate without, but worth a shot.

Hmm. I honestly don't know how much I've changed. I know there's something called usbipd in WSL, that allows devices to pass through, but you actively have to select those devices. It's used for flashing other chips in my case. See if you can at least install that and still not replicate it. Remember to terminate the running "OS" with `wsl -t {distro name}` (that you can find with `wsl -l -v`)

You are using WSL 2? `wsl --set-default-version 2`

jonasem, 2022-04-20:
Checked that they were stopped, then started ppk profile, then stopped ubuntu with `wsl -t Ubuntu`, but couldn't manage to reproduce it still :/ I installed both distros through the cli. 

```
> wsl -l -v
  NAME                   STATE           VERSION
* docker-desktop-data    Stopped         2
  Ubuntu                 Running         2
  docker-desktop         Stopped         2
  Debian                 Stopped         2
  ```
  
  I haven't installed any tooling related to usbipd, have you? In such a case I could try that as well.

mortenmoulder, 2022-04-20:
I do have usbipd installed yes. It's not passing the PPK through, tho'. https://docs.microsoft.com/en-us/windows/wsl/connect-usb

Make sure your Linux kernel is up to date as per the prerequisite.

jonasem, 2022-04-20:
So, after updating the kernel, installing usbipd and attaching the ppk, I got an error `LIBUSB error is detected.`. 

It also lists an error:
```
Error while probing usb device at bus.address 1.19: undefined. Please check that a libusb-compatible kernel driver is bound to this device, see https://github.com/NordicSemiconductor/pc-nrfconnect-launcher/blob/master/doc/win32-usb-troubleshoot.md
```
But I gather that you havent attached the ppk itself to a running wsl distribution? For me everything still works as long as I haven't attached the ppk to wsl.

mortenmoulder, 2022-04-20:
Hmm yeah, I don't attach the PPK to WSL.

Does your PPK still work even when you get that error? Or do you get that error when you try to attach it to WSL?

jonasem, 2022-04-20:
I only get the error upon attaching to wsl, after detaching it works fine again.

mortenmoulder, 2022-04-20:
Right okay.

Hmm that's so weird. I'm not even sure where I can begin to debug this issue. One thing is for sure: It's really annoying.

jonasem, 2022-04-20:
If you update to the latest nRF for Desktop, you can enable "Verbose Logging" in the About Pane in the PPK app. Clicking the "Open Log File" button after PPK disconnects might give some more information, but other than that I can't think of other things to debug it at the moment.

mortenmoulder, 2022-04-20:
I will give that a try asap and come back with a result!

mortenmoulder, 2022-04-27:
Hi again @jonasem and sorry for the delay. Been really busy. I stumbled upon a weird issue tonight. WSL didn't immediately disconnect the PPK, and I was able to disconnect and connect, enable power and start logging data after opening WSL.

However, after unplugging and plugging the PPK back in, I was no longer able to do that. Now it's just stuck at "Opening device" again. I'm using latest Power Profiler and I've enabled verbose logging. Search for `Opening device` and you can find all the times I tried to connect to the device. 3 times successfully, then I reconnected the device, and then it failed 2 times.

Here is the log:

```
2022-04-27T19:24:47.164Z DEBUG Application data folder: C:\Users\morte\AppData\Roaming\nrfconnect\pc-nrfconnect-ppk
2022-04-27T19:24:47.252Z INFO Using nrf-device-lib-js version: 0.4.5
2022-04-27T19:24:47.252Z INFO Using nrf-device-lib version: 0.11.0
2022-04-27T19:24:47.252Z INFO Using nrfjprog DLL version: 10.15.1
2022-04-27T19:24:47.252Z INFO Using JLink version: JLink_V7.58b
2022-04-27T19:24:47.268Z VERBOSE [nrfdl][trace](21:24:47.209) DeviceLister::enumerate: Checking future_status of enumerate_task_status
2022-04-27T19:24:47.268Z VERBOSE [nrfdl][trace](21:24:47.209) DeviceLister::enumerate: Waiting for enumerate task to complete
2022-04-27T19:24:47.269Z VERBOSE [nrfdl][trace](21:24:47.209) DeviceLister::enumerate::enumerate_task: Start
2022-04-27T19:24:47.270Z VERBOSE [nrfdl][trace](21:24:47.209) DeviceLister::enumerate::enumerate_task: Device_pool is empty
2022-04-27T19:24:47.271Z VERBOSE [nrfdl][trace](21:24:47.209) DeviceLister::enumerate::enumerate_task: Before async_lister->enumarate
2022-04-27T19:24:47.271Z DEBUG [nrfdl][debug](21:24:47.209) [->] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB
2022-04-27T19:24:47.272Z DEBUG [nrfdl][debug](21:24:47.209) Hub device path: \\?\USB#VID_05E3&PID_0610#7&185F5F7A&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}, port number: 4, origin: {A0A3A1F5-8837-5EBF-B44E-REDACTED}
2022-04-27T19:24:47.272Z DEBUG [nrfdl][debug](21:24:47.209) Starting DeviceIoControl
2022-04-27T19:24:47.272Z DEBUG [nrfdl][debug](21:24:47.209) Getting config descriptor
2022-04-27T19:24:47.277Z DEBUG [nrfdl][debug](21:24:47.209) Iterating descriptors.
2022-04-27T19:24:47.286Z DEBUG [nrfdl][debug](21:24:47.210) [<-] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB [took 0ms]
2022-04-27T19:24:47.287Z DEBUG [nrfdl][debug](21:24:47.210) [->] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB
2022-04-27T19:24:47.287Z DEBUG [nrfdl][debug](21:24:47.210) Hub device path: \\?\USB#VID_0BDA&PID_5411#6&61B518E&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}, port number: 3, origin: {9CD22EBE-8C7E-11EC-861D-REDACTED}
2022-04-27T19:24:47.288Z DEBUG [nrfdl][debug](21:24:47.210) Starting DeviceIoControl
2022-04-27T19:24:47.288Z DEBUG [nrfdl][debug](21:24:47.210) Getting config descriptor
2022-04-27T19:24:47.289Z DEBUG [nrfdl][debug](21:24:47.210) Iterating descriptors.
2022-04-27T19:24:47.289Z DEBUG [nrfdl][debug](21:24:47.210) [<-] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB [took 0ms]
2022-04-27T19:24:47.289Z DEBUG [nrfdl][debug](21:24:47.210) [->] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB
2022-04-27T19:24:47.289Z DEBUG [nrfdl][debug](21:24:47.210) Hub device path: \\?\USB#VID_05E3&PID_0610#7&185F5F7A&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}, port number: 2, origin: {83C1D78F-3746-5FB6-99B1-REDACTED}
2022-04-27T19:24:47.290Z DEBUG [nrfdl][debug](21:24:47.210) Starting DeviceIoControl
2022-04-27T19:24:47.290Z DEBUG [nrfdl][debug](21:24:47.210) Getting config descriptor
2022-04-27T19:24:47.291Z DEBUG [nrfdl][debug](21:24:47.210) Iterating descriptors.
2022-04-27T19:24:47.291Z DEBUG [nrfdl][debug](21:24:47.210) [<-] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB [took 0ms]
2022-04-27T19:24:47.291Z DEBUG [nrfdl][debug](21:24:47.210) [->] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB
2022-04-27T19:24:47.292Z DEBUG [nrfdl][debug](21:24:47.210) Hub device path: \\?\USB#VID_0BDA&PID_5411#7&185F5F7A&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}, port number: 4, origin: {E0AF8796-A9ED-5356-9B3C-86B3A1B88DC9}
2022-04-27T19:24:47.292Z DEBUG [nrfdl][debug](21:24:47.210) Starting DeviceIoControl
2022-04-27T19:24:47.292Z DEBUG [nrfdl][debug](21:24:47.211) Getting config descriptor
2022-04-27T19:24:47.293Z DEBUG [nrfdl][debug](21:24:47.211) Iterating descriptors.
2022-04-27T19:24:47.293Z DEBUG [nrfdl][debug](21:24:47.211) [<-] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB [took 0ms]
2022-04-27T19:24:47.293Z DEBUG [nrfdl][debug](21:24:47.211) [->] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB
2022-04-27T19:24:47.293Z DEBUG [nrfdl][debug](21:24:47.211) Hub device path: \\?\USB#VID_0BDA&PID_5411#6&61B518E&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}, port number: 4, origin: {7688366D-D8B6-5A82-AC4D-REDACTED}
2022-04-27T19:24:47.294Z DEBUG [nrfdl][debug](21:24:47.211) Starting DeviceIoControl
2022-04-27T19:24:47.294Z DEBUG [nrfdl][debug](21:24:47.212) Getting config descriptor
2022-04-27T19:24:47.294Z DEBUG [nrfdl][debug](21:24:47.212) Iterating descriptors.
2022-04-27T19:24:47.294Z DEBUG [nrfdl][debug](21:24:47.212) [<-] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB [took 0ms]
2022-04-27T19:24:47.295Z DEBUG [nrfdl][debug](21:24:47.212) [->] NRFDL::Win32::USBPlatformAPI::copyDeviceToUSB
2022-04-27T19:24:47.295Z DEBUG [nrfdl][debug](21:24:47.212) Hub device path: \\?\USB#ROOT_HUB30#4&13C2F499&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}, port number: 2, origin: {00000000-0000-0000-FFFF-FFFFFFFFFFFF}
[... cut ...]
2022-04-27T19:26:07.056Z VERBOSE [nrfdl][trace](21:26:07.047) DeviceLister::enumerate: Found device with serial number: !REDACTED!
2022-04-27T19:26:07.056Z VERBOSE [nrfdl][trace](21:26:07.047) DeviceLister::enumerate: device registered to device list
2022-04-27T19:26:07.056Z VERBOSE [nrfdl][trace](21:26:07.047) DeviceLister::enumerate: Completed waiting for enumerate task
2022-04-27T19:26:07.057Z VERBOSE [nrfdl][trace](21:26:07.047) DeviceLister::enumerate: done
2022-04-27T19:26:07.057Z WARN [nrfdl][warning](21:26:07.050)  NRFDLException(22): device event callback is already registered
```

wlgrd, 2022-05-03:
Do you have a dmesg trace when plugging in the PPK after WSL is started?

Decee1, 2023-06-25:
This has also been reproduced by me.
In my case I've seen it also having problems where i do automatic measurements in Python and then out of nowhere it stops working.
I can have my Python script measure data for 30 seconds, do something else and then measure again and then it has problems with sending data. It receives a Write timeout error from pyserial 

Something worth mentioning: 
There is no confirm sent when sending commands so there could be some problems caused by commands overlapping

mortenmoulder, 2023-06-25:
Thanks @Decee1. This is still an issue and it's really, really annoying. None of my other serial devices experience this. I'm at the point where I barely use my device, because it happens so frequently. Having to restart my computer every time it fails to communicate is a horrible experience.

Decee1, 2023-06-25:
@mortenmoulder Agree.
I am doing some more debugging on it, I have had luck with just opening the program, select device, disconnect from device and then open it again and that dissolves my connection issues. 
I just tried implementing the RESET command in my open routine hoping that would fix it. 
Else I am gonna try checking for garbage in the input buffer. Maybe 1 sample arrived too slow and then fills up the buffer leaving the PPK2 unable to communicate

Decee1, 2023-06-25:
Another update.
The RESET command works, it is just.. Sensitive?
I can reproduce problems with the device when just opening/closing it
And if i then implement a reset before closing it, it seems like it runs way more stable so it may actually be related to the Python library.

mortenmoulder, 2023-06-25:
@Decee1 I am not sure which RESET command you're referring to. I am just using the Power Profiler program.

I do have it connected to a hub most of the time, but even connected straight to my motherboard, it doesn't work consistently.

Decee1, 2023-06-25:
@mortenmoulder Oh I thought you used it though some Python script.. My bad
I really expected their program to be stable.

I found one thing which is the device reset after closing the serial communication
NOT smart imo.. 
So if for some reason the device has a bug causing it to think USB is disconnected it then resets and who knows how their Software handles that. 

I've just spend a long time trying different things out, and my best solution until now is after closing the communication then i wait for the port to disappear and wait for it to reappear.

Decee1, 2023-06-27:
Update time;
After implementing a potential fix which was waiting patiently for the device to reset whenever closing serial it was sadly not a fix.

It is pretty hard to reproduce the issue in the interface. I have tried to use the Python API to reproduce it and I haven't had any luck. I have looped my testcase over 100 times and haven't had any luck reproducing it there.
After extending the testcase a bit 
https://github.com/IRNAS/ppk2-api-python

Only in the testcases that I have made for my company I have managed to reproduce that after executing 2 testcasses and then execute a RESET testcase then it fails with reset but if I then 

And then in the actual current consumption testcase i used to be able to reproduce it but after handling the reset doing close of serial it was harder to reproduce but still possible.

**Next test**
I just tried to implement a testcase that waits x amount of minutes with no communication and after that starts communicating.
A theory I have is USB suspend suspending the device causing it to reset

wlgrd, 2023-06-28:
I asked for a dmesg while plugging in/out the PPK2 quite some time ago. Can you provide this?

Decee1, 2023-06-28:
@wlgrd I don't use Linux so in my case I cannot provide that
But if I can provide something else I please let me know. I really want this issue resolved

wlgrd, 2023-06-28:
The main issue was that this didn't work when WSL was started. Aren't you using that?

Decee1, 2023-06-28:
Nope, running Windows 10 Version 22H2

Decee1, 2023-06-28:
I don't think WSL is the issue but rather a way to reproduce the issue.
Since I've managed to reproduce communication issues with some of my testcases I think it is the same issue but just from another angle.

Decee1, 2023-07-06:
@wlgrd Is there some data i could provide you or try to do in order to troubleshoot the issue?

wlgrd, 2023-07-06:
I don't think it is the same issue, since there have been no other complaints about this issue using Windows. I have PPK2's, I have the software and I am unable to reproduce unfortunately.

Decee1, 2023-07-06:
Oh well in case it isn't I will create a new issue, on top of that I will try to see if i can reproduce the issue using the https://github.com/IRNAS/ppk2-api-python and then I can attach the testcases I made with it

wlgrd, 2023-07-06:
If you have only used the official software, I suggest opening a ticket on Nordic's DevZone would be best. If using this repository, then you can continue a new issue here =)