nordicsemi/pc-nrfconnect-ppk #284: Understanding adc value computation in Ampere Mode

sforconi, 2022-09-12:
Hi, 
I'm looking at the sources of the nrfconnect-ppk to automate some internal power consumption procedures on our boards.

The software porting is almost complete, but I don't understand how the `current_vdd` parameter is set when `PowerMode` is set to `Ampere Mode`. 

The nRF-Connect PPK interface hides the voltage slider when I switch to `Ampere Mode`, so I'm not able to set the VIN voltage with nRF-Connect.

But, looking at the source code, I can't understand which value is used as `current_vdd` when the `adc` value is converted in `getAdcResult`  (in `SerialDevice.ts`).
When working in `Ampere Mode`, the power is provided externally through PPK2 VIN pin, does the PPK2 measure the input voltage from VIN?

Could you explain more about this?

BR
Samuele.

aadnekar, 2022-09-15:
Hello @sforconi, hopefully I can help you.

You may already know the difference between Source Meter and Ampere Meter, but just to make sure we are on the same page let's review the difference.

In **Source Meter Mode** the Device under test (DUT) is supplied by the PPK, and therefore it makes sense that the PPK (with the help of the app) can set the voltage supply to the DUT.
In **Ampere Meter Mode** the DUT is supplied by some external source, for example the USB connected to your computer, which makes it (hard/impossible) for us to set the voltage supply in the app.

For your use case it sounds best to measure the voltage by using a multimeter and then set the value accordingly in your code.

We, unfortunately, do not automatically detect the voltage supply in Ampere Meter Mode. Whilst reading the code to research your question I was made aware that currently the *vdd* in Ampere Meter Mode is set by the value set in Source Meter Mode. For PPK2 it is set to 5000 by default, but when the value is changed in Source Meter Mode it is persisted. This is probably not a good thing, so I will need to investigate this some more.

Thanks a lot for reaching out, and I hope that my answer is of any help. Don't hesitate to reach out again.

sforconi, 2022-09-15:
Hi @aadnekar , 
thanks for your response.

Measuring the voltage provided to the board (that is the voltage that goes into VIN pin of PPK2) and manually set the `current_vdd` parameter, was one of the solution that we also have identified. That's ok.

My question arise from the fact that in the Nordic Power Profile software, when I select Ampere Mode, the UI hides the voltage bar, so I thought that maybe the PPK2 was able to measure the VIN voltage.

From what you have written, maybe the Power Profile UI interface should be reviewed, because it's not clear that in Ampere Meter Mode the voltage used to perform the ADC conversion was the one set in the Source Meter Mode.

Thanks again for the informations and for yous support.

BR
Samuele.

aadnekar, 2022-09-15:
Happy to be of help @sforconi.

Yes, I absolutely agree, it's not good that the value set in the Source Meter mode is used in the Ampere Meter mode, as it's not intuitive at all. Thanks for helping us identify this issue.

I'll close the issue as it's seems like you are on top of things 😄 
Thanks again.

closed 2022-09-15 by aadnekar