nordicsemi/pc-nrfconnect-ppk #291: PPK 2 not detected after 3.5.0 update

napent, 2022-09-30:
Power Profiler v3.5.0
Ubuntu 22.04

PPK 2 is not discovered after the update.
I get 'No supported devices connected' message while looking for a device.
nrf52 dk is available on the list when connected.

Rolling back to version 3.4.3 fixed the issue.
The same setup works on Windows.
 
```
# nRFConnect System Report - 2022-09-30T15-25-00.693Z

- System:     Dell Inc. Dell G15 5511
- BIOS:       Dell Inc. 1.16.0
- CPU:        1 x Intel Gen Intel® Core™ i7-11800H 2.3 GHz 16 cores (8 physical)
- Memory:     910 MB free of 15.4 GB total
- Filesystem: /dev/nvme0n1p7 (ext4) 196.9 GB 62.3% used

- OS:         Ubuntu (22.04.1 LTS) linux x64

- Versions
    - kernel: 5.15.0-48-generic
    - git: 2.34.1
    - node: 14.16.0
    - python: 3.10.6
    - python3: 3.10.6
    - nrf-device-lib-js: 0.4.13
    - nrf-device-lib: 0.12.8
    - nrfjprog DLL: 10.16.0
    - JLink: JLink_V7.70e

- Connected devices:
    - C90915813A81 : /dev/ttyACM0
    - A505MREV : /dev/ttyUSB0

```..

terjekbela, 2022-10-03:
I'm having probably the same issue with Power Profiler v3.5.0 on Ubuntu 22.04
PPK 2 is not discovered at all. This is my first time to install Profiler, my PPK2 arrived Today, so I have no experience with previous versions. Maybe I'll try 3.4.3 as @napent mentioned...

I'm getting these errors on the log:
---------------------------------------------------------------------------------
00:15:18.824	Using nrf-device-lib-js version: 0.4.13
00:15:18.824	Using nrf-device-lib version: 0.12.8
**00:15:18.824	Unable to detect version of nrfjprog DLL.
00:15:18.824	Unable to detect version of JLink.**

---------------------------------------------------------------------------------
# nRFConnect System Report - 2022-10-03T22-17-28.624Z

- System:     LENOVO 20QV000WHV
- BIOS:       LENOVO N2OET42W (1.29 )
- CPU:        1 x Intel Core™ i7-9750H 2.6 GHz 12 cores (6 physical)
- Memory:     3.2 GB free of 15.3 GB total
- Filesystem: /dev/nvme1n1p2 (ext4) 1.8 TB 10.8% used

- OS:         Ubuntu (22.04.1 LTS) linux x64

- Versions
    - kernel: 5.15.0-48-generic
    - git: 2.34.1
    - node: 14.16.0
    - python: 
    - python3: 3.10.6
    - nrf-device-lib-js: 0.4.13
    - nrf-device-lib: 0.12.8
    - nrfjprog DLL: Unknown
    - JLink: Unknown

- Connected devices:

datenreisender, 2022-10-05:
Ubuntu 22 is not yet generally supported for nRF Connect for Desktop, but I do not know whether this really causes problems here. Please ask these questions rather on https://devzone.nordicsemi.com, there are more experienced support folks over there and you will get quicker help than here.

RaoulSP, 2022-10-12:
Hi, just wanted to let you know that the issue has been fixed. It seems that v.3.5.0 tried to filter out devices that were not PPK2's, but was a little too strict.

v.3.5.2 undoes the filtering, and is now available in nRF Connect for Desktop.

Related case on DevZone: https://devzone.nordicsemi.com/f/nordic-q-a/92623/power-profiler-kit-ii-not-visable-on-power-profiler-v3-5-0

napent, 2022-10-20:
It is resolved with the newest update.-

closed 2022-10-26 by datenreisender