nordicsemi/pc-nrfconnect-ppk #296: Logarithmic scale

danielkucera, 2022-11-27:
Hello,
the device is great!
However, would it be possible to make the Y-axis (with current) logarithmic? 
My device has a wide range of consumption so it would be easier to read uA values when there is also 100mA  on the same scale.
[image]

danielkucera, 2022-11-27:
I was able to change the axis but I am not that pro in js to be able to make the button work. Can you please help?

[image]

my code is in #297

danielkucera, 2022-11-27:
I was able to change the axis but I am not that pro in js to be able to make the button work. Can you please help?
Some hint is here: https://stackoverflow.com/questions/42625579/dynamically-update-the-scale-type-of-chartjs-graph

[image]

my code is in #297

aadnekar, 2023-03-08:
PR is merged: https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/pull/297

closed 2023-03-08 by aadnekar