nordicsemi/pc-nrfconnect-ppk #320: Is it possible to get the Voltage level in ampere meter mode ?

yguel, 2023-04-18:
Reading the schematics of the PPK2, it looks like the voltage of the source (VDUT SENSE) is available to the PPK2.
Would-it be possible to get its value in ampere meter mode ?

wlgrd, 2023-04-19:
In theory, yes. It is used in a loopback when setting the output voltage, but it can also be read during operation. Unfortunately, this is not implemented but is in our backlog. 
It would demand some sacrifices to the readout speed of the current measurement. How often would you think you would need to have the voltage of the DUT updated?

yguel, 2023-04-19:
For our application it is 10Hz, quite low compared to the usual current measurement needs.

yguel, 2023-04-19:
Is that part of the code open source ? We can dedicate ressources to implement the feature.

wlgrd, 2023-04-19:
Unfortunately not, this demands a firmware change. I will get back to you when time allows for prioritizing this

KSt-HW, 2023-08-04:
We would need the same, but with a higher frequency. We would like to have it with each current reading, since we are having a very dynamic source and sink as well.
That would probably mean, that the possible sampling speed is at least half the current possible speed, but that would be ok with us.
I addition, it would be handy for the UI, if an area of the measured graph is selected, not only the current and charge are calculated, but also the power by including the measured voltages.

tmarkson, 2023-08-04:
Count me as another user that's OK with lower sampling speed when reading Voltage.  It's a very enticing feature!  I think the only major feature missing from PPK ecosystem.

frolswe, 2024-02-11:
Voltage measurement would be really useful, please implement that :)
Any possibility for you to opensource the firmware for this device?

oldgerman, 2024-04-24:
The current waveform graph shows the current, which is compensated for by calculations using calibration data, as this assumes that the voltage is constant, then when the voltage fluctuates more, the less it reflects the true current. I hope the new firmware will add the ability to display both current and voltage waveforms at the same time, and to switch between the display of these two types of data :)
1: Constant voltage + real time current 
2: real-time voltage + real-time current

oldgerman, 2024-04-24:
The current waveform graph shows the current, which I presume is compensated for by calculations using calibration data, and if the voltage is assumed to be constant in the calculations, the less it reflects the true current when the voltage fluctuates a lot. I hope the new firmware will add the ability to display current and voltage waveforms at the same time, and to switch between the display of these two types of data :)
1: Constant voltage + real time current
2: real-time voltage + real-time current

MM-Lehmann, 2024-09-24:
That would be a very useful feature, actually mitigating the most serious disadvantage over other power meters, as indicated by other users above. We could imagine using this device in a number of projects then.
Any chance, you could open source or at least provide the firmware for the controller? We (Fraunhofer IIS) could dedicate some resources implementing the feature in firmware and feed that back to you.

JarrettR, 2025-07-09:
Bumping this to try and get it moved up the roadmap. It would be a killer feature that would keep me invested in this ecosystem

Beny-Compteq, 2026-01-29:
I second this. For tests of a real battery life the voltage measurement would be very useful thing :)

JarrettR, 2026-09-28:
Mentioning it in here too, because folks might be interested:
I built a fork that does this:
https://github.com/JarrettR/Joule-Counter