nordicsemi/pc-nrfconnect-ppk #329: [Feature] Is it possible to allow sampling periods greater than 500 seconds?

Kampi, 2023-08-29:
I guess the sampling period is set only in the UI and not in the firmware? So would it be possible to add the option for a custom period?

jonasem, 2023-08-29:
It is only set in the UI, the maximum duration is dependent on what sampling rate you use. It is theroretically possible to add a custom period option.

aadnekar, 2023-08-30:
Hello @Kampi 
We do allow longer sampling periods, ⚠️⚠️ just be very cautious⚠️⚠️, because we do store the whole active sample in-memory. This means that performance may be significantly reduced, and you may encounter unexpected errors. So if you do choose to do longer samples, don't increase the buffer by too much.

You can access the advanced settings after selecting device, and pressing `Shift+Alt+Ctrl+A`, and then change the Max size of buffer.

We do want to improve this in the future, but we do not have any short term plan for it.

Hope this helps, and feel free to reach out again. I will close this, as we do have it in our backlog, but feel free to reopen.

closed 2023-08-30 by aadnekar

Kampi, 2023-08-30:
Hi @aadnekar 

thanks for the explanation. You helped me a lot :)