nordicsemi/pc-nrfconnect-ppk #374: Power Profiler current measurements showing 0 on 100ms scale.

markb-EE, 2023-12-18:
If the Power Profiler app, when the 100 ms scale is selected, the average, max and charge boxes all read 0.

[image]

datenreisender, 2023-12-20:
I was unable to reproduce this. If this was also done with 4.0.0-beta1: Could you first check whether this problem still exists with the latest version? Does it only happen at 100ms, not above or below? Could you export a run to a .ppk file and share it?

KonstKonovalov, 2023-12-29:
Average calculation is still incorrect in version v.4.0.0-beta3. The average current cannot be greater than the max current.
[image]
[ppk-20231229T142044.zip](https://github.com/NordicSemiconductor/pc-nrfconnect-ppk/files/13794204/ppk-20231229T142044.zip)

datenreisender, 2024-01-03:
Ah, yes, I am now able to reproduce this with your PPK file by navigating to one of the peaks and then zooming in. Seems the further I zoom in, the further the values are off. 

Also interesting that in your image the location in the minimap is wrong, since it shows you to be left of the peak.

Thanks for the PPK file!

@kylebonnici: Do you have an idea what causes this? 

[image]

kylebonnici, 2024-01-03:
Issue has been found and a PR will follow soon the issue is that the average is off by a factor of 1000 as we have change the unit in the chart and omit changing unit in the Window average box

[image]

kylebonnici, 2024-01-03:
Will look at the other issue where values are zeros

kylebonnici, 2024-01-04:
Hi @markb-EE and @KonstKonovalov version 4.0.0-beta4 is out and should address these issues

closed 2024-01-04 by kylebonnici