nordicsemi/pc-nrfconnect-ppk #516: PPK2 "no device setup found" issue

sacca97, 2025-05-02:
I can see my PP2K, but when I tried to connect, it just doesn't. I am on Fedora.
Here's the log:

```
2025-05-02T12:29:03.904Z WARN Required component nrf-udev is not detected. Install it from https://github.com/NordicSemiconductor/nrf-udev and restart the application
2025-05-02T12:29:04.044Z DEBUG Sending event "ppk: running nrfutil device"
2025-05-02T12:29:04.169Z DEBUG Sending event "ppk: running nrfutil device"
2025-05-02T12:29:04.170Z DEBUG Sending event "ppk: running nrfutil device"
2025-05-02T12:29:04.175Z DEBUG Sending event "ppk: running nrfutil device"
2025-05-02T12:29:04.178Z INFO Using the bundled core version for nrfutil device: 8.0.0
2025-05-02T12:29:04.368Z INFO Getting serial port options from the persistent store for 001050234808.pc-nrfconnect-ppk
2025-05-02T12:29:04.368Z INFO Getting serial port options from the persistent store for 000683533264.pc-nrfconnect-ppk
2025-05-02T12:29:04.369Z INFO Getting serial port options from the persistent store for DA0372FC5550.pc-nrfconnect-ppk
2025-05-02T12:29:04.369Z INFO Device Connected SN:001050234808
2025-05-02T12:29:04.371Z DEBUG Sending event "ppk: device connected"
2025-05-02T12:29:04.371Z INFO Device Connected SN:000683533264
2025-05-02T12:29:04.371Z DEBUG Sending event "ppk: device connected"
2025-05-02T12:29:04.372Z INFO Device Connected SN:DA0372FC5550
2025-05-02T12:29:04.372Z DEBUG Sending event "ppk: device connected"
2025-05-02T12:29:04.386Z INFO Using nrfutil-device version: 2.6.4
2025-05-02T12:29:04.386Z INFO Using nrf-device-lib version: 0.17.35
2025-05-02T12:29:04.386Z INFO Using nrfjprog DLL version: 10.24.2
2025-05-02T12:29:04.386Z INFO Using JLink version: JLink_V8.24
2025-05-02T12:29:04.386Z WARN Installed JLink version does not match the expected version (JLink_V7.94i)
2025-05-02T12:29:09.073Z INFO Selecting device with the serial number DA0372FC5550
2025-05-02T12:29:09.090Z DEBUG Sending event "ppk: running nrfutil device"
2025-05-02T12:29:09.396Z INFO Selected device with the serial number DA0372FC5550
2025-05-02T12:29:09.396Z INFO Validating firmware for device with s/n DA0372FC5550
2025-05-02T12:29:09.397Z INFO Connected to device with the serial number DA0372FC5550 and family: Unknown 
2025-05-02T12:29:09.397Z INFO Note: no pre-compiled firmware is available for the selected device. 
2025-05-02T12:29:09.397Z ERROR Error while setting up device with the serial number DA0372FC5550
2025-05-02T12:29:09.397Z ERROR "No device setup found"
2025-05-02T12:29:09.398Z INFO Deselected device
2025-05-02T12:29:09.398Z INFO Deselecting device
2025-05-02T12:29:09.398Z DEBUG Sending event "ppk: device selected"
2025-05-02T12:29:09.399Z DEBUG Sending event "ppk: device deselected "
```

vogon, 2025-09-24:
yep!

vogon, 2025-10-07:
I've been using my PPK2 under Windows in a virtual machine for the past couple weeks and after reflashing the firmware it's worked fine; out of curiosity I tried it under Linux again tonight and it appears that now that the firmware has been reflashed, Linux is able to access it just fine as well.  for anyone else having this issue, it might be worth installing Windows and nRF Connect in a VM, and following the steps in this thread to manually reflash the firmware on your PPK2: https://devzone.nordicsemi.com/f/nordic-q-a/113563/ppk2-firmware-update-done-incorrectly